STMicroelectronics/STM32CubeH5 #2: Missing DSP libraries

Makin-Things, 2023-07-23:
**Describe the set-up**
 * The board (either ST RPN reference or your custom board).
   Nucleo-STM32H563ZI
 * IDE or at least the compiler and its version.
   STM32CubeIDE 1.13.0

**Describe the bug**
The firmware pack is missing the precompiled maths libraries. For other MCU's they are in Drivers/CMSIS/Lib or Drivers/CMSIS/DSP/Lib, but for the H5 they are missing. Can you please add these to the repository or provide instructions on how to build from source.

TOUNSTM, 2023-08-02:
Hello @Makin-Things,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With Regards,

Makin-Things, 2023-08-17:
@TOUNSTM is there any update on this issue?
Are you also able to point me in the direction of being able to raise issues and get support? The company I work for is willing to pay if needed, but getting timely responses is critical to the project we are working on.

TOUNSTM, 2023-08-29:
Hello @Makin-Things,

Thank you again for your contribution. Unfortunately, the pre-compiled DSP libraries are no longer provided by arm in the CMSIS package. This change occurred in `CMSIS v5.8.0`, so it’s up to you to add the necessary sources from `CMSIS/DSP/Source` to your project.

You can check this [Link](https://www.keil.com/pack/doc/CMSIS/DSP/html/index.html#using) it can be helpful. I will keep you informed if there are any updates that may help you.

With regards,

Makin-Things, 2023-09-10:
Hello @TOUNSTM ,

Thanks for the update. I had noticed that ARM had split the DSP into it's own repo.
Are there any instructions on how to get it compiling from source? I have tried and failed badly. It would also be nice if it was properly incorporated into STM32CubeIDE/CubeMX, there should simply be a an option within the Computing section to activate the DSP code.

Cheers
Simon

ALABSTM, 2024-03-07:
See also [cmsis_core#5](https://github.com/STMicroelectronics/cmsis_core/issues/5)

ALABSTM, 2024-04-01:
ST Internal Reference: 175542

TOUNSTM, 2024-06-13:
Hello @Makin-Things,

I hope you are doing well. We finally got the feedback of our development teams.
ST was not responsible for generating the DSP libraries; instead, these were obtained directly from the official CMSIS Core release. Unfortunately, there is no plan to deliver DSP libraries at this time.

I hope this answers your point. please allow me to close this issue.

With regards,

closed 2024-06-13 by TOUNSTM