STMicroelectronics/STM32CubeH5 #3: OB_BOR_LEVEL_x: probable error in stm32h5xx_hal_flash_ex.h file

alessandro-ce, 2023-07-25:
Hi,
I think that the macros OB_BOR_LEVEL_1, OB_BOR_LEVEL_2 and OB_BOR_LEVEL_3 are wrong. They are defined in the STM32H5xx_HAL_Driver _stm32h5xx_hal_flash_ex.h_ header file, starting from line 305.

According to page 328 of the [reference manual](https://www.st.com/resource/en/reference_manual/rm0481-stm32h563h573-and-stm32h562-armbased-32bit-mcus-stmicroelectronics.pdf) of the STM32H5, their values should be the following:

`#define OB_BOR_LEVEL_1        (FLASH_OPTSR_BOR_LEV_1 | FLASH_OPTSR_BOR_LEV_0)  /*!< Reset level 1 threshold */`
`#define OB_BOR_LEVEL_2        FLASH_OPTSR_BOR_LEV_0                            /*!< Reset level 2 threshold */`
`#define OB_BOR_LEVEL_3        FLASH_OPTSR_BOR_LEV_1                            /*!< Reset level 3 threshold */`

Thank you
Alessandro

RJMSTM, 2023-07-28:
Hello @alessandro-ce,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,
Rania

TOUNSTM, 2023-08-02:
ST Internal Reference: 158618

KRASTM, 2024-03-21:
Fixed in 182f4c8589bf775b330ed63af7c0c17060d8a649

closed 2024-03-21 by KRASTM