STMicroelectronics/STM32CubeH5 #4: System clock configuration issue in STM32H573I-DK OEMiROT_Boot sample project

Saman-Faghih-1988, 2023-08-02:
Hello,
It seems that there is an issue in the PLL1M configuration in _Projects/STM32H573I-DK/Applications/ROT/OEMiROT_Boot/Src/system_stm32h5xx.c_ file _SetSysClock()_ function. **RCC_PLL1CFGR_PLL1M_Pos** was used as the value.
https://github.com/STMicroelectronics/STM32CubeH5/blob/3d723b2467cdb13c2d780ceaa583b269680b67a4/Projects/STM32H573I-DK/Applications/ROT/OEMiROT_Boot/Src/system_stm32h5xx.c#L298

Also, I think the values for reaching 250MHz should be like this:
- PLLM: 4 (RCC_PLL1CFGR_PLL1M_Pos will expand to 8)
- PLLN: 31 (it's 60 in the current code)
- PLLP: 2
- PLLQ: 2
- PLLR: 2

**Screenshot**
[image]

Furthermore, it might be better to use:
_RCC_PLL1_VCORANGE_WIDE_ instead of _RCC_PLL1VCOWIDE_
_RCC_PLL1_VCIRANGE_2_ instead of _RCC_PLL1CFGR_PLL1SRC_1_
_RCC_SYSCLKSOURCE_PLLCLK_ instead of _(RCC_CFGR1_SW_1 | RCC_CFGR1_SW_0)_

https://github.com/STMicroelectronics/STM32CubeH5/blob/3d723b2467cdb13c2d780ceaa583b269680b67a4/Projects/STM32H573I-DK/Applications/ROT/OEMiROT_Boot/Src/system_stm32h5xx.c#L297

HBOSTM, 2023-08-04:
ST Internal Reference: 158775

ALABSTM, 2025-02-28:
Hi @CuriousSaman,

I hope you are doing well. Just to let you know this issue has been solved in the frame of version **1.5.0**, released just now.

Thank you for your patience and thank you again for having reported.

With regards,

closed 2025-02-28 by ALABSTM