STMicroelectronics/STM32CubeH5 #5: STM32CubeIde generates incorrect code for I2C4

Makin-Things, 2023-08-17:
**Describe the set-up**
 * NUCLEO-H563ZI
 * STM32CubeIDE 1.13.1
 * STM32Cube MCU Package for STM32H5 Series 1.1.0

**Describe the bug**
When you enable Event or Error interrupts in the NVIC controller the generated source code in stm32h5xx_it.c is incorrect. If you enable the error interrupt you get this.

```
/**
  * @brief This function handles I2C4 ERROR interrupt.
  */
void I2C4_EV_IRQHandler(void)
{
  /* USER CODE BEGIN I2C4_EV_IRQn 0 */

  /* USER CODE END I2C4_EV_IRQn 0 */
  HAL_I2C_EV_IRQHandler(&hi2c4);
  /* USER CODE BEGIN I2C4_EV_IRQn 1 */

  /* USER CODE END I2C4_EV_IRQn 1 */
}
```

Note that the comment is correct, but the function name is that of the EVENT interrupt.  Enabling the I2C4 EVENT interrupt results in the inverse of this.

**How To Reproduce**
1. Create a new project based on the H5 series MCU and enable I2C4 and then enable the ERROR/EVENT interrupts in the NVIC.
2. Generate the code.
3. Observe that the code is incorrect in the generated stm32h5xx_it.c source file.

**Additional context**
This is not a problem with I2C1 and I2C2 (I haven't checked I2C3 as I can't enable it in my project because of used resources).

HBOSTM, 2023-08-24:
ST Internal Reference: 159911

HBOSTM, 2023-08-24:
Hi @Makin-Things ,

Thank you for this contribution, this point has been reported to our development teams. I will get back to you as soon as I have any updates.

With regards,

HBOSTM, 2023-08-24:
Hi @Makin-Things ,

The point you reported is actually related to the CubeMX generation problem and not to the firmware published in this repository. Unfortunately, we don't treat aspect related to CubeMX tool in our GitHub repositories. They are rather treated at the [STM32CubeMX](https://community.st.com/s/topic/0TO0X000000BTr8WAG/stm32cubemx) dedicated page of the ST Community ,However it has been reported.

Since this issue is not directly related to the STM32Cube firmware but rather to our ecosystem, please allow me then to close this thread. Thank you for your comprehension.

With regards,

closed 2023-08-24 by HBOSTM