STMicroelectronics/STM32CubeH5 #6: LAN8742 driver will always find PHY on device address 31

jotrax, 2023-08-18:
**Describe the set-up**
NUCLEO-H563ZI
CubeIDE 1.13.0 and CubeMX 6.9.0 using STM32Cube FW_H5 V1.1.0

**Describe the bug**
In the function LAN8742_Init(..) the PHY address is determined by polling the SMR with different addresses.

```
     /* Get the device address from special mode register */
     for(addr = 0; addr <= LAN8742_MAX_DEV_ADDR; addr ++)
     {
       if(pObj->IO.ReadReg(addr, LAN8742_SMR, &regvalue) < 0)
       {
         status = LAN8742_STATUS_READ_ERROR;
         /* Can't read from this device address
            continue with next address */
         continue;
       }

       if((regvalue & LAN8742_SMR_PHY_ADDR) == addr)
       {
         pObj->DevAddr = addr;
         status = LAN8742_STATUS_OK;
         break;
       }
     }

     if(pObj->DevAddr > LAN8742_MAX_DEV_ADDR)
     {
       status = LAN8742_STATUS_ADDRESS_ERROR;
     }
```

_addr_ won't get larger than 31 but later the address is checked if larger than 31 (LAN8742_MAX_DEV_ADDR) with will never lead to a LAN8742_STATUS_ADDRESS_ERROR.

HBOSTM, 2023-08-24:
Hello @jotrax ,

Thank you for this contribution.
I can't reproduce the problem from my side. So could you please give us more details about how you got this issue? And share the steps you have followed to reproduce this problem.

With regards,

jotrax, 2023-08-29:
The way how the address detection is implemented will never return with LAN8742_STATUS_ADDRESS_ERROR.

You can test it e.g. by removing the solder bridges SB62 and/or SB69. This simply cuts the MDIO connection to the PHY so the initial device address scan must not work. I would expect LAN8742_Init() to return with LAN8742_STATUS_ADDRESS_ERROR, but instead the function assumes the PHY to be found on address 31 and fails a few codes lines later with LAN8742_STATUS_RESET_TIMEOUT.

Reading the BCR register and requiring Bit 15 (Soft-Reset) to be cleared in addition (or instead) to reading the SMR register allows to detect if there was no PHY found --> LAN8742_STATUS_ADDRESS_ERROR.

ASEHSTM, 2024-08-16:
Hello @jotrax,

Sorry for the late reply. 
Before the loop, pObj->DevAddr is initialized to `LAN8742_MAX_DEV_ADDR + 1`, which is greater than 31. After the device address detection loop, if no valid address is found within the loop, `pObj->DevAddr` will remain at its initial value. Consequently, the condition `if (pObj->DevAddr > LAN8742_MAX_DEV_ADDR)` will be true, and the status will be correctly set to LAN8742_STATUS_ADDRESS_ERROR.

With Regards,

ASEHSTM, 2025-03-24:
Hello @jotrax,

Please allow me to close this thread.

With Regards,

closed 2025-03-24 by ASEHSTM