STMicroelectronics/STM32CubeH5 #8: Missing link template files for MDK-ARM

xu-yingjing, 2023-08-24:
Hello

Will link template files be added in the path of "Drivers\CMSIS\Device\ST\STM32H5xx\Source\Templates\arm"? Otherwise, how do I allocate NSC space for the Secure project in MDK-ARM? Is it referring to STM32L5? By the way, the TrustZone project for STM32H5 generated using STM32CubeMX does not seem to specify a location for NSC in Flash, but is automatically allocated by MDK, is this a problem?

Thank you

TOUNSTM, 2024-05-29:
ST Internal Reference: 182803

TOUNSTM, 2024-05-31:
Hello @xu-yingjing,

First we would like to thank you for your report.

>  **1-** Will link template files be added in the path of "Drivers\CMSIS\Device\ST\STM32H5xx\Source\Templates\arm"? 

Actually, the point you raised has been reported to our development teams. We will get back to you as soon we have updates.

> **2-** how do I allocate NSC space for the Secure project in MDK-ARM? Is it referring to STM32L5? By the way, the TrustZone project for STM32H5 generated using STM32CubeMX does not seem to specify a location for NSC in Flash, but is automatically allocated by MDK, is this a problem?`

This point is related to the CubeMX generation problem and not directly related to the firmware published in this repository. Unfortunately we don't treat aspect related to CubeMX tool in our GitHub repositories.
In case you have CubeMX-related issues to share, please submit them in the [ST Community](https://community.st.com/s/question/0D53W00000JhsU4SAJ/stm32cubemx-ide-issue-throws-warning-when-using-macros) in the CubeMX related section. There you shall find people who will address your request and whom you will hopefully find a solution with.

Thank you for your comprehension and thank you again for having reported.

Kind regards,

closed 2024-06-03 by xu-yingjing