STMicroelectronics/STM32CubeH5 #13: EEPROM emulation problem on STM32H563ZI

Bhuvi-Naveen, 2023-11-05:
Hi,

1. I'm facing problem in reading data from EEPROM, which was written before power cycle.
2. EE_Init() always causes formatting of EEPROM irrespective of EE_CONDITIONAL_ERASE or EE_FORCED_ERASE parameter passed, resulting in erasing the previous data

**Configuration:** "STM32H5 series using high-cycle data area" with EDATA_ENABLED  &  START_PAGE_ADDRESS      0x0900C000U

I'm using the latest version of EEPROM emulation driver "X-CUBE-EEPROM version-6.0.0" on dev kit "NUCLEO-H563ZI"

The example code works fine, it is able to write and read the data from the flash every time on power up by formatting the page, but when I wanted to read the data which was written in previous cycle, It always reads data zero.

The reason I found was,  it always give current page status as "ERASING". 

I made the following change in function FindPage(), having a switch case operation for "FIND_READ_PAGE" by adding a check currentpagestatus == STATE_PAGE_VALID as shown below. 

I added **STATE_PAGE_VALID** check instead of **STATE_PAGE_ERASING**, because, the debugger shows the current page status as valid in function FindPage()

case FIND_READ_PAGE:  /* ---- Read operation ---- */
 if (currentpagestatus == STATE_PAGE_ACTIVE || **currentpagestatus == STATE_PAGE_VALID)**
 {
        return currentpage;
 }
 else
 {
        if (currentpagestatus == STATE_PAGE_RECEIVE)
        {
          return previouspage;
        }
        else
        {
          return EE_NO_PAGE_FOUND;   /* No active Page */
        }
}

In main.c I added the following code just before initializing the EE_Init
**ConfigureCrc();**
for (Index = 1; Index < NB_OF_VARIABLES+1; Index++)
 {
      EE_ReadVariable32bits(Index, &a_VarDataTab[Index-1]);
 }
with the above change I'm now able to read data which was written in previously.

Am I missing some settings? please help!

TOUNSTM, 2024-06-05:
ST Internal Reference: 183453

TOUNSTM, 2024-06-06:
Hello @Bhuvi-Naveen,

Actually, the point you raised has already been answered in our [Community Forum](https://community.st.com/t5/stm32-mcus-embedded-software/eeprom-emulation-problem-on-stm32h563/m-p/604991#M43090) . And as our software does not require any fixes at the moment, please allow me to close this thread.

Thank you very much for your attention. We look forward to hearing from you.

With regards,

closed 2024-06-06 by TOUNSTM