STMicroelectronics/STM32CubeH5 #14: STM32H563/573: HAL_SPI)TransmitReceive not working on 1.8V

trembel, 2024-01-31:
**Describe the set-up**
 * NUCLEO-H563ZI, (NUCLEO563ZI$MR2, STM32H563ZIT6U AA327 VQ X TWN AA 316 06)
 * STM32CubeIDE, Version: 1.14.0, Build: 19471_20231121_1200 (UTC)
 * GNU Tools for STM32 (11.3.rel1)
 * Jumper JP4 set to 1.8V

**Describe the bug**
SPI transaction (on both SPI1 and SPI2) does not work correctly when the MCU is running at 1.8V. When using `HAL_SPI_TransmitReceive` with `HAL_MAX_DELAY`, then the function never returns. With 3.3V it works as expected.

**How To Reproduce**
See attached CubeIDE project
[test_spi.zip](https://github.com/STMicroelectronics/STM32CubeH5/files/14109641/test_spi.zip)

**Additional context**
After some debugging, it seems that the `SPI_FLAG_RXP` (`stm32h5xx_hal_spi.c:1383`) is never set. Thus, the while loop inside `HAL_SPI_TransmitReceive` (`stm32h5xx_hal_spi.c:1367`) will never return.

KRASTM, 2024-02-07:
Hello @trembel,

Thank you for this contribution.

First, I agree with you that `HAL_SPI_TransmitReceive ` API does not work correctly when the MCU running at 1.8V and the SPI Clock set at 250 Mhz.
But after analyzing the issue, you should put your SPI clock at 120 Mhz according to the datasheet [DS14121 Rev2 ](https://www.st.com/resource/en/datasheet/stm32h573ai.pdf)
[image]

Also, there is a configuration that must be followed.
[image]

Best Regards
Karim

KRASTM, 2024-02-19:
Hello @trembel,

Any News?
Did you manage to get it to work?

BRs

KRASTM, 2024-02-19:
Hello @trembel,

I'm glad that you manage to resolve the issue.

Please, allow me to close the issue.
BRs

trembel, 2024-02-20:
Would there be a way to enforce a CubeMX warning on both
- to high SPI frequency
- to low drive settings for the selected speed (which I couldn't find a reference for in the DS)
I think this could help to mitigate such issues directly.