STMicroelectronics/STM32CubeH5 #15: Compilation of cmsis_os2.c fails if USE_MEMORY_POOL_ALLOCATION is defined.

Vinzenz82, 2024-07-25:
**Describe the set-up**
 STM32CubeIDE Version: 1.15.1

**Describe the bug**
```
#ifdef USE_DYNAMIC_MEMORY_ALLOCATION
  unused_memory = (CHAR *)_tx_initialize_unused_memory;
#elif  USE_MEMORY_POOL_ALLOCATION
  static CHAR freememStack[RTOS2_BYTE_POOL_STACK_SIZE + RTOS2_INTERNAL_BYTE_POOL_SIZE];
  static CHAR freememHeap[RTOS2_BYTE_POOL_HEAP_SIZE + RTOS2_INTERNAL_BYTE_POOL_SIZE];
  unused_memory_Stack = (CHAR *)freememStack;
  unused_memory_Heap = (CHAR *)freememHeap;
#endif
```

Compiler Output:
`cmsis_os2.c:270:34: error: #elif with no expression`

**How To Reproduce**
define USE_MEMORY_POOL_ALLOCATION instead of USE_DYNAMIC_MEMORY_ALLOCATION

2. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...).
stm32h5/Middlewares/ST/cmsis_rtos_threadx/cmsis_os2.c

TOUNSTM, 2024-07-29:
ST Internal Reference: 187607

TOUNSTM, 2024-07-29:
See Also https://github.com/STMicroelectronics/stm32_mw_cmsis_rtos_tx/issues/2

TOUNSTM, 2025-09-16:
Fixed in 0d526a4930310cf8fa6dce7538ab891cc37eb250