STMicroelectronics/STM32CubeH5 #17: STM32H5 -> Cmosis-M33 S/NS issue

ArkadiuszRaj, 2025-01-31:
**Caution**
The file cortex_cm33.h defines only secure version of SCB.

https://github.com/STMicroelectronics/STM32CubeH5/blob/d4c01dd10d43ebf90d631d199a1a304bd8ffb84e/Drivers/CMSIS/Include/core_cm33.h#L2208C1-L2209C1

**Describe the bug**
SCB-VTOR sets VTOR_S, no possibility to set VTOR_NS

TOUNSTM, 2025-02-07:
Hello @ArkadiuszRaj,

You point looks relevant. It has been forwarded to our development teams. I will keep you updated.

Best regards,

TOUNSTM, 2025-02-12:
Hello @ArkadiuszRaj,

Thank you for your report. Unfortunately, your point is not entirely clear. These files are provided by Arm, and we use them as they are. Additionally, I would like to point out that `SCB_NS` is declared at [line 2244](https://github.com/STMicroelectronics/STM32CubeH5/blob/d4c01dd10d43ebf90d631d199a1a304bd8ffb84e/Drivers/CMSIS/Include/core_cm33.h#L2244C11-L2244C17) if that is what you are referring to. Could you please provide more details?

Kind regards,

closed 2025-03-12 by TOUNSTM