STMicroelectronics/STM32CubeH5 #18: Feature Request: User Defined Context pointer in HAL Handles

jostlowe, 2025-03-14:
Hi! 

I would like to re-open an old issue from the STM32CubeG4 repo:

https://github.com/STMicroelectronics/STM32CubeG4/issues/23

Hoping that there might be better luck here in the H5 repo :)

Short version:
It would be really beneficial to have a `void*` pointer in the HAL handles to associate any context that a user might need, which can be accessed in e.g. callbacks. This could make writing modular code and e.g. middleware simpler.

For details and discussions, see the attached issue in the G4 repo.

This feature could can potentially be enabled and disabled in config in the same way as registrable callbacks to prevent bloat.

Hoping this is something you could look into. Cheers!

koniarik, 2025-03-17:
Yes, I concur that this would be most beneficial and the fact that we did not even get it for G4 is sad, H5 would definetly benefit from this too :)

Until that the registrable callbacks themselves lack usability for us, and hence we opted to not usem them at all, use the plain old direct callback and just setup custom registration mechanism.

jostlowe, 2025-03-18:
Here's a community post linking to several similar discussions:

https://community.st.com/t5/stm32-mcus-embedded-software/user-defined-context-in-hal-callbacks/m-p/783093#M61168

ALABSTM, 2025-04-09:
Hi @jostlowe and @koniarik,

The feature you requested has been confirmed to be in the pipe and should be part of a future release. However, this will not be for the short term. We cannot share a date for the moment. I will keep you informed.

Please allow me to close this issue for the moment. Thank you for your comprehension and thank you again for having tackled this topic.

With regards,

**PS:** I set the label `wontfix` simply to signify the change request is not considered yet **the moment** the issue has been closed at. But, as mentioned above, the feature is definitely in the pipe.

closed 2025-04-09 by ALABSTM