STMicroelectronics/STM32CubeH5 #19: Can you give a simple example of IWDG with HAL_IWDG_RegisterCallback?

25th-engineer, 2025-03-24:
I am trying to implement an IWDG in my program, which will save the value of the specified register to EEPROM before resetting. <br>
But I found that my program is not working. <br>
I politely request your developers to provide a simple example of IWDG using HAL_SWDG_Registercallbacks. <br>
Many thanks. :-)
```c
void IWDG_EarlyWakeupCallback(IWDG_HandleTypeDef *hiwdg)
{
    uint8_t arr[1] = {0xAA};
    // if (imageAbortedFlag)
    {
        if (getCurrentRunningImage() == 0x01)
        {
            pRAMbootctrl->image_A_tasks_aborted_count++;
        }
        else if (getCurrentRunningImage() == 0x10)
        {
            pRAMbootctrl->image_B_tasks_aborted_count++;
        }
        MemoryStoreData(EEPROM_MSATABLE_UP03_INDEX, UPE6_128, 1, arr);
        jump_to_IMAGE(BOOTLOADER_START_ADDRESS);
    }
}

void MX_IWDG_Init(void)
{
    hiwdg.Instance = IWDG;
    hiwdg.Init.Prescaler = IWDG_PRESCALER_8;
    hiwdg.Init.Window = IWDG_WINDOW_DISABLE;
    hiwdg.Init.Reload = 0x0FFF; // ?TBD
    hiwdg.Init.EWI = 0; // ?? 0x0FFE

    if (HAL_IWDG_Init(&hiwdg) != HAL_OK)
    {
        // Error_Handler();
    }

    if (HAL_IWDG_RegisterCallback(&hiwdg, HAL_IWDG_EWI_CB_ID, IWDG_EarlyWakeupCallback) != HAL_OK)
    {
        // Error_Handler();
    }

    #if 0
    HAL_NVIC_SetPriority(IWDG_IRQn, NVICPreemptionPriority_IWDG, NVICSubPriority_IWDG);
    HAL_NVIC_EnableIRQ(IWDG_IRQn);
    #endif
}
```

RJMSTM, 2025-03-28:
Hello @25th-engineer,

Unfortunately, I tried to find an example of IWDG with HAL_IWDG_RegisterCallback for you, but there isn't one in our firmware. All examples are handled with an interrupt when the push-button is pressed.

Regards,

ALABSTM, 2025-04-24:
Hi @25th-engineer,

As pointed out by @RJMSTM, there is no IWDG example using the user callback registration. However, there are two applications provided with the STM32CubeU5 firmware:
* [TCP Echo Client](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/NetXDuo/Nx_TCP_Echo_Client) running on the **B-U585I-IOT02A** board,
* [I2C Tx-Rx with LPBAM](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/NUCLEO-U575ZI-Q/Applications/LPBAM/LPBAM_I2C_TransmitReceive) running on the **NUCLEO-U575ZI-Q** board.

You will notice such lines in the [stm32u5xx_hal_conf.h](https://github.com/STMicroelectronics/STM32CubeU5/blob/2f30bf3950ee49ff0314631d160e22c2367dda06/Projects/NUCLEO-U575ZI-Q/Applications/LPBAM/LPBAM_I2C_TransmitReceive/Core/Inc/stm32u5xx_hal_conf.h#L211) files to **enable** the user callbacks registration...

```c
#define  USE_HAL_I2C_REGISTER_CALLBACKS        1U /* I2C register callback enabled       */
#define  USE_HAL_RTC_REGISTER_CALLBACKS        1U /* RTC register callback enabled       */
```

... and, depending on the application, the user callbacks **defined**...

```c
/* USER CODE BEGIN Slave_ReloadSeq_Config */

/**
  * @brief  i2c3 error callback
  * @param  hi2c3 pointer to I2C handle
  * @retval None
  */
static void I2C3_Error_Callback(I2C_HandleTypeDef *hi2c3)
{
  /* Turn LED3 on */
  BSP_LED_On(LED3);
}
/* USER CODE END Slave_ReloadSeq_Config */
```

... and **registered** in files such as the [lpbam_slave_reloadseq_config.c](https://github.com/STMicroelectronics/STM32CubeU5/blob/2f30bf3950ee49ff0314631d160e22c2367dda06/Projects/NUCLEO-U575ZI-Q/Applications/LPBAM/LPBAM_I2C_TransmitReceive/LPBAM/Slave/lpbam_slave_reloadseq_config.c#L549-L553), as show below...

```c
  /* Register i2c3 error callback */
  if (HAL_I2C_RegisterCallback(&hi2c3, HAL_I2C_ERROR_CB_ID, I2C3_Error_Callback) != HAL_OK)
```

I hope this helps.

With regards,

25th-engineer, 2025-04-28:
> Hello [@25th-engineer](https://github.com/25th-engineer),
> 
> Unfortunately, I tried to find an example of IWDG with HAL_IWDG_RegisterCallback for you, but there isn't one in our firmware. All examples are handled with an interrupt when the push-button is pressed.
> 
> Regards,

@RJMSTM Thank you for your explanation. :-)

25th-engineer, 2025-04-28:
> Hi [@25th-engineer](https://github.com/25th-engineer),
> 
> As pointed out by [@RJMSTM](https://github.com/RJMSTM), there is no IWDG example using the user callback registration. However, there are two applications provided with the **STM32CubeU5** firmware:
> 
> * [TCP Echo Client](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/B-U585I-IOT02A/Applications/NetXDuo/Nx_TCP_Echo_Client) running on the **B-U585I-IOT02A** board,
> * [I2C Tx-Rx with LPBAM](https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Projects/NUCLEO-U575ZI-Q/Applications/LPBAM/LPBAM_I2C_TransmitReceive) running on the **NUCLEO-U575ZI-Q** board.
> 
> You will notice such lines in the [stm32u5xx_hal_conf.h](https://github.com/STMicroelectronics/STM32CubeU5/blob/2f30bf3950ee49ff0314631d160e22c2367dda06/Projects/NUCLEO-U575ZI-Q/Applications/LPBAM/LPBAM_I2C_TransmitReceive/Core/Inc/stm32u5xx_hal_conf.h#L211) files to **enable** the user callbacks registration...
> 
> #define  USE_HAL_I2C_REGISTER_CALLBACKS        1U /* I2C register callback enabled       */
> #define  USE_HAL_RTC_REGISTER_CALLBACKS        1U /* RTC register callback enabled       */
> ... and, depending on the application, the user callbacks **defined**...
> 
> /* USER CODE BEGIN Slave_ReloadSeq_Config */
> 
> /**
>   * @brief  i2c3 error callback
>   * @param  hi2c3 pointer to I2C handle
>   * @retval None
>   */
> static void I2C3_Error_Callback(I2C_HandleTypeDef *hi2c3)
> {
>   /* Turn LED3 on */
>   BSP_LED_On(LED3);
> }
> /* USER CODE END Slave_ReloadSeq_Config */
> ... and **registered** in files such as the [lpbam_slave_reloadseq_config.c](https://github.com/STMicroelectronics/STM32CubeU5/blob/2f30bf3950ee49ff0314631d160e22c2367dda06/Projects/NUCLEO-U575ZI-Q/Applications/LPBAM/LPBAM_I2C_TransmitReceive/LPBAM/Slave/lpbam_slave_reloadseq_config.c#L549-L553), as show below...
> 
>   /* Register i2c3 error callback */
>   if (HAL_I2C_RegisterCallback(&hi2c3, HAL_I2C_ERROR_CB_ID, I2C3_Error_Callback) != HAL_OK)
> I hope this helps.
> 
> With regards,

@ALABSTM Hi, Thanks. I'd fixed my problem but I would still like to say a big thank to you. I believe that your explanation will help other developers in the future. :-)

closed 2025-04-28 by RJMSTM

ALABSTM, 2025-04-28:
Hi @25th-engineer,

You're very welcome. Really glad to read you are now able to register your callbacks.

With regards,