STMicroelectronics/STM32CubeH5 #20: warning: 'node_info.currentnode_addr' may be used uninitialized when using DMA with -Ofast compile flag

Halogenek, 2025-03-26:
**Describe the set-up**
Custom board with STM32H523ZWTx
IDE: STM32CubeIDE 1.18.0
STM32Cube MCU Package and version: STM32H5 1.5.0

**Describe the bug**
When I try to use DMA with -Ofast flag i got the following warning during compilation

```
../Drivers/STM32H5xx_HAL_Driver/Src/stm32h5xx_hal_dma_ex.c: In function 'HAL_DMAEx_List_ConvertQToDynamic':
../Drivers/STM32H5xx_HAL_Driver/Src/stm32h5xx_hal_dma_ex.c:2992:77: warning: 'node_info.currentnode_addr' may be used uninitialized [-Wmaybe-uninitialized]
 2992 |     if (((uint32_t)pQList->FirstCircularNode != 0U)                         &&
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~
 2993 |         ((uint32_t)pQList->FirstCircularNode != node_info.currentnode_addr) &&
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  
../Drivers/STM32H5xx_HAL_Driver/Src/stm32h5xx_hal_dma_ex.c:2939:26: note: 'node_info.currentnode_addr' was declared here
 2939 |   DMA_NodeInQInfoTypeDef node_info;
      |                          ^~~~~~~~~
```

**How To Reproduce**
1. Create a blank C code ST project for STM32H523ZET6 with library copy.
2. Enable USART1 as Asynchronous with Data Direction: Receive Only - other parameters default.
3. Enable GPDMA1 Channel 0 - 2 Words internal FIFO as Standard Request Mode
4. Configure the GPDMA1 Channel 0 as Circular Mode: Enable, Request: USART1_RX, Destination Address Increment After Transfer: Enabled - other parameters default.
5. In NVIC enable USART1 global interrupt.
6. Save and generate code - when prompted for enabling cache - ignore and click Yes.
7. Right click on your project and open the projects Properties. Go to C/C++ Build, Settings, Tool Settings, MCU/MPU GCC Compiler, Optimization and use Optimization level: Optimize for speed (-Ofast). Click Apply and Apply and Close.
8. Right click your project and click on Clean Project.
9. Click on the build icon.

**Attachments**
I added the exported file to this issue [dma-test.zip](https://github.com/user-attachments/files/19472574/dma-test.zip)

Halogenek, 2025-03-26:
I am experiencing some DMA hiccups in my device with this configuration, but the device is part of a complex system so I am still investigating if those are related. Problems like DMA is not working on first initialization and random lockups.
Anyway the search lead me to this warning, which should not exist. :)

KRASTM, 2025-03-28:
Hello @Halogenek,

Thank you for the report.

I get the same warning with your project. I will check with our team and run some tests, and I will keep you posted.

With regards