STMicroelectronics/STM32CubeH5 #21: FIXME should be fixed or removed in #define OEMIROT_OB_BOOT_LOCK OB_BOOT_LOCK_ENABLE

jespertragardh, 2025-05-16:
In `Applications/ROT/OEMiROT_Boot/Inc/boot_hal_cfg.h` for projects NUCLEO-H533RE, NUCLEO-H563ZI and STM32H573I-DK
```C
#ifdef OEMIROT_DEV_MODE
#define OEMIROT_OB_BOOT_LOCK OB_BOOT_LOCK_ENABLE /*!< BOOT Lock expected value  */
#define OEMIROT_ERROR_HANDLER_STOP_EXEC /*!< Error handler stops execution (else it resets) */
#else
#define OEMIROT_OB_BOOT_LOCK OB_BOOT_LOCK_ENABLE /*!< FIXME BOOT Lock expected value  */
#endif /* OEMIROT_DEV_MODE */
```

The FIXME doesn't seem relevant. OB_BOOT_LOCK_ENABLE (defined in stm32h5xx_hal_flash_ex.h) is 0xB4U which according to RM0481 means enabled so you won't be able to change SECBOOT_ADD or SWAP_BANK which makes the boot setup more secure.

But since this is security related SW, any FIXME comment causes some unease in code security review.

**Request**
Please either fix what needs to be fixed or remove the FIXME.

KRASTM, 2025-05-20:
ST Internal Reference: 210144

KRASTM, 2025-09-26:
Fixed in Commit: https://github.com/STMicroelectronics/STM32CubeH5/commit/86360fcf2e3c8b3eb0b774957ee8fd707a6b65cf