STMicroelectronics/STM32CubeH5 #22: Redundant files OEMiROT_Boot/STM32CubeIDE/Example/Startup/startup_stm32h5*.s

jespertragardh, 2025-05-21:
**Describe the set-up**
 * STM32CubeIDE 1.18.0

**Additional context**
The files `Projects/NUCLEO-H533RE/Applications/ROT/OEMiROT_Boot/STM32CubeIDE/Example/Startup/startup_stm32h533retx.s` and `Projects/NUCLEO-H563ZI/Applications/ROT/OEMiROT_Boot/STM32CubeIDE/Example/Startup/startup_stm32h563zitx.s` (they do not seem to exist for the other boards covered by this git) are redundant.

They contain a reset handler and interrupt vectors but these are never actually linked. The linker script in `OEMiROT_Boot/STM32CubeIDE/stm32h5xx_bl2.ld` links the implementations from `OEMiROT_Boot/Src/startup_stm32h5xx.c` which in turn partly relies on `_mainCRTStartup` from `lib/gcc/arm-none-eabi/13.3.1/thumb/v8-m.main+fp/hard/crt*.o` of the tool chain.

I assume these are left-overs from creating a new STM32CubeIDE project and then setting up the project to use the common OEMiROT and mcuboot code.

They confuse the customers trying to debug the Reset_Handler.

I suggest that they be removed from the project's sources.

KRASTM, 2025-05-26:
ST Internal Reference: 210776

KRASTM, 2025-09-26:
Fixed in Commit: https://github.com/STMicroelectronics/STM32CubeH5/commit/160c89558d312274f822ace0a83ed1b33516fd86

closed 2025-09-26 by KRASTM